Uvic-Robotics-Club/uvic-rover #41: [6-Feature] Gazebo simulation with the arm

JoshuaGabriel, 2023-09-17:
**Is your feature request related to a problem? Please describe.**
Get the arm on gazebo

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

JoshuaGabriel, 2023-09-23:
An <inertia> element within each <link> element must be properly specified and configured.

We might need to calculate this individually for each link. I haven't tried using gazebo yet with the arm, but this might be needed according to the docs on here. (http://classic.gazebosim.org/tutorials?tut=ros_urdf&cat=connect_ros)

https://docs.google.com/document/d/12b684tOl-aLRubx9HFht0qc5BPMtknjQJH7_3LxKOOE/edit

JoshuaGabriel, 2023-09-25:
If someone can turn this into a xacro file that would be great

http://wiki.ros.org/urdf/Tutorials/Using%20Xacro%20to%20Clean%20Up%20a%20URDF%20File

JoshuaGabriel, 2023-09-25:
Haven't been able to get the arm into gazebo yet, this is where I left off. Might have something to do with having no 'dummy' link, not sure yet. 
https://answers.ros.org/question/268655/gazebo_ros_control-plugin-is-waiting-for-model-urdf-in-parameter/

this would be a lot easier with xacro

This tutorial helps, need to launch the world 
`roslaunch gazebo_ros empty_world.launch`
then run 
```
rosrun gazebo_ros spawn_model -file `rospack find uvic_rover`/rover_description/UVicArm.urdf -urdf -x 0 -y 0 -z 1 -model MYROBOT
```

chanhee20park, 2023-10-08:
To launch the arm in gazebo, run
`source ~/catkin_ws/devel/setup.bash
roslaunch uvic_rover robot_xacro.launch`

There are still two issues to fix:
-The models that load in are too big
-When unpaused, the different parts fall apart

JoshuaGabriel, 2023-10-19:
Look into this, for getting moveit installed
https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html

JoshuaGabriel, 2023-10-19:
I got the moveit package and tutorials to work if I build the moveit packages with `catkin build`, I had to delete the devel and build folder, then used catkin build

closed 2024-05-29 by JoshuaGabriel